ReturnInfinity/BareMetal-OS-legacy #75: e1000 driver does not setup bus mastering

mntmn, 2015-02-24:
Hi there!

Disclaimer: this is one of my first forays into x86 driver land, so I probably got something wrong. 

I was trying to get networking in BareMetal-OS to work under QEMU using -device e1000-82544gc. The card was correctly probed by the PCI code and initialised by the code in i8254x.asm, but I couldn't send or receive packets. I debugged e1000.c in QEMU with lots of printfs and noticed that in start_xmit(), pci_dma_read() would complete but read only zeroes from BareMetal's os_eth_tx_buffer address. I then discovered that e1000_can_receive() would always return 0 because PCI_COMMAND_MASTER in s->parent_obj.config[PCI_COMMAND] was not set.

Googling turned up this patch to QEMU http://lists.gnu.org/archive/html/qemu-devel/2014-12/msg02818.html which disables the check for the bus mastering flag on this device. And voila, it works. My question is now: is it possible to correctly set up bus mastering in BareMetal for a PCI device?

Thanks for any feedback.

IanSeyler, 2015-02-24:
I've uploaded a potential fix for this. Can you please give it a try?

mntmn, 2015-02-25:
Cool, that was fast! I just merged your fix locally and can now use networking with an unpatched QEMU. So it seems to work, thank you very much!

closed 2015-02-25 by mntmn

IanSeyler, 2015-02-25:
Awesome! I'm glad that did the trick. Would you mind sharing how you start up QEMU with networking? I'd like to test and add it to the documentation. I've only tested the network code in VirtualBox or with physical hardware.

mntmn, 2015-02-25:
Sure:

    qemu-system-x86_64 -monitor stdio -vga std -smp 2 -m 256 -device ahci,id=ahci -drive id=disk,file=bmfs.image,if=none -device ide-drive,drive=disk,bus=ahci.0 -name "BareMetal OS" -netdev bridge,id=net0  -device e1000-82544gc,netdev=net0,mac=52:54:de:ad:be:ef,id=nd0

IanSeyler, 2015-02-25:
Thanks again!