ReturnInfinity/BareMetal-OS-legacy #92: CRLF or LF or CR or ?

Roxxik, 2015-07-07:
What exactly is the newline convention at the moment?

I porsanally think we should just adopt the *nix convention[1] here, because I think most of the stuff that gets compiled for BM is done from a *nix environment.

[1] copy-paste from https://en.wikipedia.org/wiki/Newline
    LF:    Multics, Unix and Unix-like systems (Linux, OS X, FreeBSD, AIX, Xenix, etc.), BeOS, Amiga, RISC OS and others.[1]
    CR:    Commodore 8-bit machines, Acorn BBC, ZX Spectrum, TRS-80, Apple II family, Mac OS up to version 9 and OS-9
    RS:    QNX pre-POSIX implementation.
    0x9B:  Atari 8-bit machines using ATASCII variant of ASCII. (155 in decimal)
    LF+CR: Acorn BBC and RISC OS spooled text output.
    CR+LF: Microsoft Windows, DEC TOPS-10, RT-11 and most other early non-Unix and non-IBM OSes, CP/M, MP/M, DOS (MS-DOS, PC DOS, etc.), Atari TOS, OS/2, Symbian OS, Palm OS, Amstrad CPC

ohnx, 2015-07-07:
yeah, most of us here are using Unix/Linux when contributing to code, so LF is probably the way to go.

IanSeyler, 2015-07-07:
LF is used in all source code since I develop mainly in Linux.

Documentation may be CR+LF since I write in Windows sometimes.

closed 2015-07-07 by IanSeyler

Roxxik, 2015-07-07:
because os_output_chars handles CR too, and my assembly isn't that good so i'm not quite sure what happens there

IanSeyler, 2015-07-07:
Yep, it should be able to handle both LF and CR+LF formats.