ReturnInfinity/BareMetal-OS-legacy #94: os_output_char in Kernel API

Roxxik, 2015-07-07:
There is currently no way to just put one character on screen.
To prepare the new syscall we need to be able to output an arbitary char first. The current output functions only allow to output a printable char or space or a whole string

For that i recommend a few changes to the screen.asm

rename os_output_char to os_output_printable
create new function os_output_char
change os_output_chars to repeatedly call os_output_char

the interface of os_output_char is the same as the interface of os_output_printable
os_output_char either prints newline/tab or calls os_output_printable

after this has been done the os_output_char could be part of the Kernel API.

ohnx, 2015-07-07:
I believe this has changed in the other version of BareMetal OS [here](https://github.com/ReturnInfinity/BareMetal-kernel).

Roxxik, 2015-07-07:
Haven't seen that yet, why are there two versions of the OS?

Roxxik, 2015-07-07:
eventhough in the other repo is no syscall that can output a newline directly

IanSeyler, 2015-07-07:
os_output_chars is able to print a single character. In needs the address of where the character is in memory and a count value of 1.

To print a newline you can use this:

mov rsi, newline
mov rcx, 1
call os_output_chars
...
newline:		db 13, 0

closed 2015-07-07 by IanSeyler

Roxxik, 2015-07-07:
i'd do this like
char c = '\n';
b_output_chars(&c,1);