ReturnInfinity/BareMetal-OS-legacy #97: BareMetal Rust PoC

Roxxik, 2015-07-07:
just take a look
https://github.com/Roxxik/Hello-BareMetal-Rust
i try and improve this further to get arbitary rust code running

IanSeyler, 2015-07-13:
Slick! I haven't had a chance to try it but looks good. Does Rust support inline assembly?

benaryorg, 2015-07-13:
Yes it does.

Roxxik, 2015-07-13:
I chose to not include rust inline assembly, because currently it is unstable, undocumented and not conformant to gcc inline assembly.
That's why i have included a c layer to make the systemcalls.

(I tried to use it but the compiler always optimized the assembler away :(  )

benaryorg, 2015-07-14:
@IanSeyler here, have some [inline assembly](http://doc.rust-lang.org/nightly/book/inline-assembly.html).