ReturnInfinity/BareMetal-OS-legacy #101: porting baremetal-os on risc-v

8l, 2015-07-22:
Hi, 

i am interested in porting baremetal-os on risc-v.
http://riscv.org/

ANGEL - Browser-based RISC-V ISA simulator
http://riscv.org/angel/

IanSeyler, 2015-07-23:
I'm also interested in this. https://github.com/ReturnInfinity/BareMetal-kernel was set up with a separate x86-64 folder for this reason. I'd also like to write a version for 64-bit ARM.

8l, 2015-07-23:
and here is yet another cpu, it also looks promising
different design, but implemented in chisel.
http://www.rexcomputing.com/
http://www.theplatform.net/2015/07/22/supercomputer-chip-startup-scores-funding-darpa-contract/

IanSeyler, 2016-11-18:
If there is enough demand this can go into BareMetal-kernel

closed 2016-11-18 by IanSeyler