ReturnInfinity/BareMetal-OS-legacy #105: Memory mapped files (or file mapped virtual address space one program/process/thread)

Konard, 2015-10-29:
Are any plans for this? It is actually the only critical dependency for my database/platform to be ported to BareMetal OS and I'm still stuck with Windows and Unix here.. May be it will be even more efficient not to use small pages (4KB) but instead to use (2MB) pages.

This technique allows to give an infinite memory to any program/process (that is limited only by hard drives).

IanSeyler, 2015-11-01:
Using 2MiB pages would be ideal as that is what BareMetal already uses for memory mapping.

I'll investigate how this could be added. I'm just not sure of how caching should be configured.

Konard, 2015-12-30:
May be it is a good idea to make memory mapping as a "plugin"? So the developer can deside how to optimize caching?

ohnx, 2015-12-30:
@Konard Making BareMetal-OS a modular OS, where you could select extra features is an interesting idea.

Konard, 2015-12-31:
At least we can select features (modules or packages) that should be included OS (before compiling or writing to disk/flash). It can be done through modification of code, or through plugin system. By hand or automatically.

effbiae, 2016-01-01:
Maybe, as with another famous kernel, a config file (optionally generated
by a menu system) can choose to include or exclude code.

The code might be best kept in one repo or optionally got from another
repo. (Hybrid system).

I'm almost at the point of switching to BM but being able to map disk
blocks into an address space is important for me.

Jack

IanSeyler, 2016-01-05:
Fixing the memory mapping that is currently being used is here: https://github.com/ReturnInfinity/BareMetal-kernel/issues/14

All further development is going into BareMetal-kernel. I'm moving away from a sort of "general purpose" OS to a more "server node" type of OS.

Konard, 2016-01-05:
So what repository is most ready to be used? https://github.com/ReturnInfinity/BareMetal-OS or https://github.com/ReturnInfinity/BareMetal-kernel ?

ohnx, 2016-01-05:
@Konard the [BareMetal-kernel](https://github.com/ReturnInfinity/BareMetal-kernel) is most up-to-date, but lacks many of the features present in [BareMetal-OS](https://github.com/ReturnInfinity/BareMetal-OS). However, most of the big changes in the kernel are also committed to this repository.

IanSeyler, 2016-01-06:
@Konard BareMetal-OS at the moment as there is no way to control an instance of just BareMetal-kernel yet as I stripped out the CLI.

D3nshi, 2016-06-10:
Hello,
I think you have to try this thing, it's so hilarious, take a look at <http://treltixyndo.xtra-power.net/lnpemdp>

Pardon my monkey thumbs, phil@hoefer-elze.de

norberto5757, 2016-06-11:
look with deep sincerity do not know when or I receive your e'm just a
systems programmer with a very broad curriculum right through 37 years of
living today than my children offer me forgiveness if at any time
incosiente were taxed. my data but not. nothing to disturb my social
environment much less in a country that I admire ye visited for several
years my name is norberto gonzalez rojas and my mobile in mexico is
6645872021 my address from my home in Tijuana, Baja californa is Andador A
#, 31 elRubi apologies please delete my information from your database ever
and hurt anyone thanks blessings also I inform you that do not speak
English use the Google translator only thing that 100% are programming
terms what I do on social networks is to speak God does not send me anyone
is not charging its own initiative or annoying people tendencista'm not
religious and I am apolotitico only love God above all things blessings.
Sorry wan more time

ohnx, 2016-06-11:
uhh, @IanSeyler, I think you should remove @norberto5757's comment...

También, @norberto5757, ¿qué?

ohnx, 2016-06-11:
@norberto5757, ¿qué?

El correo electrónico al que está respondiendo es "spam".

Nada de que preocuparse.

D3nshi, 2016-07-04:
Hello,

I think I've found the solution to your problem at last, read it here please <http://lespozasi.jamesmilles.com/lnrlsfw>

All best, phil@hoefer-elze.de

D3nshi, 2016-07-21:
Hi,

I've came  across that new amazing stuff, don't miss it up! Just take  a look here <http://improve.sparkallday.com/lninnw>

Hope this helps, phil@hoefer-elze.de

D3nshi, 2016-07-21:
Hi,

We haven't talked for  a  while  so I just  wanted to  say Hi  and share  with you some information,  read  more here  please <http://honour.simvision.net/lnjibrk>

Warmest, phil@hoefer-elze.de

D3nshi, 2016-07-21:
Hi,

I'm working on a new project and need your  opinion on  some issues,  please take  a look here <http://have.ssg-corp.com/lnkmj>

Wishes, phil@hoefer-elze.de

D3nshi, 2016-07-25:
Greetings! 

I'm not sure whether I should tell  you that, please read it yourself <http://please.verifydrugs.com/lnbwz>

Sent from a prehistoric stone tablet, phil

4485234

D3nshi, 2016-08-25:
Dear friend! 

What a  great news I've got for you!  Just check it  out, you're going to be surprised  <http://physlikazu.thebrownbandit.com/lnybrv>

In haste, phil

D3nshi, 2016-08-25:
Greetings, 

I  want you to cast  your mind back to the first time  we met, take  a  look at the it here  <http://chizonote.smoke-blower.com/lnrrode>

phil

D3nshi, 2016-08-25:
Dear friend! 

I  found some  cool stuff that  I couldn't pass, take a  look at  that <http://pracruphenu.lifeworksimaging.com/lnblkfgv>

phil

D3nshi, 2016-08-25:
Hey! 

I wanted to tell you that I'm doing fine,  even  better  that ever, you may find the reasons of it here <http://churdapryxa.taylorsrockinjewelry.com/lnlimx>

phil

D3nshi, 2016-08-25:
Dear friend! 

I've got  some interesting news for you, you're  gonna be pleaseantly surprised. Read more here   <http://shadaquophy.mesivta.net/lnhev>

Looking forward, phil

D3nshi, 2016-08-25:
Dear, 

I've heard about a new fish restaurant opening next  saturday, are you  planning to  go there? Here is some  info about that  <http://rochuchide.locodog.com/lnuufu>

Very truly yours, phil

D3nshi, 2016-08-25:
Hello, 

I've got some great news  since we've met last time, please read it here  <http://jihuhinda.ultimatebackyardevents.com/lnqazb>

My Best, phil

D3nshi, 2016-08-25:
Hello! 

There is something super interesting for you, just take  a  look, you'll never regret. Please, read  more here <http://froshundefry.priusliftsupports.com/lnkfz>

Sent from my iPhone, phil

D3nshi, 2016-08-25:
Hello friend, 
I found  that amazing stuff and I can tell it's the best of  its kind,  just take a look  <http://thongacypha.internetdatingmonitor.com/lnllcva>
Thx, phil