ReturnInfinity/BareMetal-OS-legacy #110: [INFO] 64-bit ARM

iceblu3710, 2016-02-10:
You mentioned "I'd also like to write a version for 64-bit ARM", have you made any progress on this yet? I am interested in running a cluster of Odroid-XU4's.  

What is the major hurdle in porting to Arm's 64bit ASM from X86_64?

scherrey, 2016-02-10:
Major hurdle is you rewrite everything from scratch. :-)

  - - Ben

ohnx, 2016-02-10:
including a bootloader (so it's not just the OS)

iceblu3710, 2016-02-12:
Well obviously, but other than that its totally a weekend project right? lol

What is the major difference between the languages, are their major x86 instructions that ASM does not have so those routines would need to be completely re-written?

I imagine if they share 80% of similar instructions the majority of the work could be mapping.

hikilaka, 2016-02-17:
The instructions are very different.

scherrey, 2016-02-18:
The CPU architecture is completely different. You can't reason about your
software and CPU state in the same way across the two CPU architectures. I
think the only similarities is that they are both large register-based CPUs
(as opposed to something a bit more esoteric like a stack-based
architecture) and have some overlapping memory access methods. Would really
be an entirely independent project.

  -- Ben

IanSeyler, 2016-02-22:
This probably won't happen in BareMetal OS. The BareMetal kernel is a possibility though: https://github.com/ReturnInfinity/BareMetal-kernel/tree/master/src/armv8

I have yet to find an ARMv8 assembler though.

And yes, it would be a total rewrite retaining the same API.

IanSeyler, 2016-11-15:
Closing this issue as it will be part of BareMetal-kernel when an ARMv8 version comes.

closed 2016-11-15 by IanSeyler