ReturnInfinity/BareMetal-OS-legacy #111: Stuck at iddrive_poll with VirtualBox

charray, 2016-04-17:
Hi,

I have pulled the latest version from git and compiled it today.  Whenever I insert an AHCI drive, The system loops indefinitely after printing `[mem: 64 MiB]   [hdd:`.  After some `call os_output` debugging, I found the system loops indefinitely on the line `jne iddrive_poll` in `os/drivers/storage/achi.asm`.

The VirtualBox I used is version 5.0.14.  I tried with various disk sizes and the issue is repeatable.  Is there anything I can do to troubleshoot?  Thanks.

IanSeyler, 2016-04-19:
I will give this a try on my end and see what I can find. Thanks for reporting this.

Moritz30, 2016-06-12:
Had the same issue and can help:
Make sure your virtual disc is an SATA one.

IanSeyler, 2016-11-14:
This is confirmed as a bug. I tested with VirtualBox 5.1.8 and got the same results.

IanSeyler, 2016-11-16:
Check "Offset 10h: PxIS – Port x Interrupt Status" to see what the issue may be.

IanSeyler, 2016-11-18:
Set AHCI Enable (AE)? Currently the driver does not do this.

IanSeyler, 2016-11-18:
It's not AE (VirtualBox already has in enabled) and all of the BIOS handoff bits are clear.

IanSeyler, 2016-11-18:
Neat! So it's just iddrive function that hangs. Comment it out and BareMetal starts up fine (although it detects the disk is 0MiB). Reading from the disk after this works.

IanSeyler, 2016-11-18:
Set the FIS length to 5. No idea why it didn't work with 4 in VirtualBox 😕 .

closed 2016-11-18 by IanSeyler