ReturnInfinity/BareMetal-OS-legacy #115: SATA drive support

nbrecht01, 2016-06-14:
BareMetal OS should support SATA disk drives. The kernel loads fine, but the disk appears empty when running the 'dir' command, saying it is 0 bytes big.

IanSeyler, 2016-06-14:
BareMetal does support SATA disks via the AHCI driver.

- What hardware are you running it on?
- Was the drive formatted with BMFS?

nbrecht01, 2016-07-13:
Sorry for taking so long to answer!
Yes, the drive was BMFS formatted.
Hardware:
         AMD Athalon 64
         M3A78-CM Motrherboard

The kernel boots fine, but the disk size is 0B and none of my apps will show up with the 'dir' command.

IanSeyler, 2016-10-06:
Check your SATA Configuration in the BIOS. "OnChip SATA Type" should be set to AHCI.

closed 2016-10-06 by IanSeyler