ReturnInfinity/BareMetal-OS-legacy #119: README.md says autoconf -fi

x86poem, 2016-10-17:
# README.md says simply "autoconf -fi" but it raise error
# Instead, following commands seems works
cd src/BMFS
sudo apt-get install automake libtool
libtoolize --force
aclocal
autoheader
automake --force-missing --add-missing
autoconf -f -i

./configure
make
sudo make install

ohnx, 2016-10-17:
I think the idea is that you typically want to use [these scripts](https://github.com/ReturnInfinity/BareMetal) to build BareMetal-OS.

Where does the README say to use `autoconf -fi`? I can't seem to find that.

x86poem, 2016-10-17:
Oh.. Sorry. I confused with BMFS.
https://github.com/ReturnInfinity/BMFS/blob/master/README.md

ohnx, 2016-10-17:
I think you should be running `autoreconf -fi`, and not `autoconf -fi`. `autoreconf` automatically calls [`aclocal`, `autoheader`, and those other commands](http://stackoverflow.com/a/27334612).

x86poem, 2016-10-17:
I just checked again.
It says "autoreconf -fi"
... sorry.

closed 2016-10-17 by x86poem