ReturnInfinity/BareMetal-OS-legacy #122: tried to add elf format support but failed to run any app

onesea, 2016-11-30:
I tried to compile the kernel to elf64 format and the kernel start and run, but it failed to run any app. The testing app code:

mem_app:
        mov rax, 0x0000c300001234b8 ; machine code for:  mov rax 0x1234 + ret
        mov rdi, 0x0000000000200000
        stosq
        call 0x0000000000200000
        ret

It is okay to call mem_app in the boot loader, but it hang up if I call mem_all right at the begin of kernel_start. any idea what's going wrong?

IanSeyler, 2016-11-30:
I don't see any problems with the code you provided - I added it the the CLI testzone on my dev system and it worked correctly. I'll need more details on this (other code changes, compile methods, etc).

ustime, 2016-12-01:
I added some code in the pure64 boot loader to load elf kernel. the kernel is then compile using:
     nasm -felf64 -o kernel64.o kernel64.asm
     ld -T kernel.ld -o kernel64.sys kernel64.o
     strip kernel64.sys
of course also added entry point declaration and removed the ORG instruction.

now after some debugging, it seemed that mmu is not functioning properly. because the instruction,
            call 0x200000
actually becomes something like
            call 0x30af....
i saw that through qemu monitor window

onesea, 2016-12-01:
I added some code in the pure64 boot loader to load elf kernel. the kernel is then compile using:

nasm -felf64 -o kernel64.o kernel64.asm
ld -T kernel.ld -o kernel64.sys kernel64.o
strip kernel64.sys

of course also added entry point declaration and removed the ORG instruction.

now after some debugging, it seemed that mmu is not functioning properly. because the instruction,
call 0x200000
actually becomes something like
call 0x30af....    ; the address had automatically changed to 3M+
i saw that through qemu monitor window. i can't see any reason to cause such issue.