ReturnInfinity/BareMetal-OS-legacy #125: Hi, I can't to run the BareMetal-OS

Eshmundev, 2017-06-21:
I  downloaded the BareMetal-kernel zip file, and after to Decompress it, I introduced the scripts ./setup, ./build.sh and run.sh , to the final of introducing the script ./run_x86-64.sh, the console it said: "qemu-system-x86_64: -drive id=disk,file=bmfs.image,if=non e,format=raw: Could not open 'bmfs.image': No such file or directory",  It's the only thing that's wrong, Why is it wrong?, I send a one picture of the console. thanks for the help!
[image]

tay10r, 2017-06-21:
Create an empty bmfs.image file and it should work just fine.

tay10r, 2017-06-21:
If you need to work with the filesystem image, use the BMFS utilities.

tay10r, 2017-06-21:
Use this to get the utilities:

```
sudo apt-add-repository ppa:taylorcholberton/bmfs-utils
sudo apt update
sudo apt install bmfs-utils
```

The command `bmfs-init` can initialize a disk image of a specific size.

These other commands may help you:

 - `bmfs-cat`
 - `bmfs-cp`
 - `bmfs-rm`

They all use the file `disk.image` by default, but you can use the `--disk` option to change that, or just rename the file.

They all support the `--help` option.

Eshmundev, 2017-06-21:
thanks for your help, I check it, especially the bmfs utils, That you are
well !.

mrm1ck, 2017-10-27:
I was going to say try master or original as I have my apps all modified for running multi system and in a sparatic output

closed 2017-11-06 by IanSeyler